MazinLab/MKIDReadout #2: R Bug

baileyji, 2018-08-16:


neelay893, 2018-08-21:
The problem was a flipped optimal filter - we thought the firmware was doing a correlation, but it was actually a convolution. 

Histogram of phase stream data optimally filtered by firmware:
[image]

Optimally filtered in software, using the same 50 tap filter loaded in firmware. R looks much better...
[image]

Optimally filtered by firmware, with a flipped filter coefficients loaded. R now agrees w/ software filtered data.
[image]

Optimally filtered in software using a flipped template (convolution).
[image]

baileyji, 2018-08-21:
Reopening until fix committed.

zobristnicholas, 2018-08-31:
Data from the 20180830 MEC run suggests that R~2 when it should be higher. The previous fix did not fully solve the problem

neelay893, 2018-09-14:
Data taken on 9/13/2018 suggests that R from analog and digital readout are consistent w/ each other. R~2 on MEC seems to be a separate issue unrelated to the readout.

commit a9e55ccc (2018-08-24): fix #2 flipped the filters to work with a convolution instead of a corralation
commit bccf4147 (2018-08-26): Updated optimal filter calculations to really fix #2
closed 2018-09-14 by neelay893