MazinLab/MKIDReadout #31: Auto-ADC atten infinite loop.

abwalter, 2018-08-25:
The function for automatically finding the ADC attenuation fell into an infinite loop. There were two boards trying to adjust their ADC attens at the same time; this may be a coincidence. 

Ideally: fix cause of infinite loop. 
Fallback: If we can't figure out why it's looping then add a loop counter and error out on too many loops.

abwalter, 2018-08-25:
It happens with just one board at a time. Here's some example output:
iRms 0.8045176285437422
qRms 0.7915580817087216
atten3 7.0
atten4 7.0
iRms 0.0358977282055029
qRms 0.035645546147516693
atten3 0.25
atten4 0.25
iRms 0.8029727811266437
qRms 0.7906611741913939
atten3 7.0
atten4 7.0
iRms 0.034695085813494314
qRms 0.03544920646782865
atten3 0.0
atten4 0.25
iRms 0.804086399039635
qRms 0.7917742110254182
atten3 6.75
atten4 7.0

neelay893, 2018-08-26:
Didn't mean to close that. One possibility is that changeAtten was not waiting for the V7 before returning, so the ADC snapshot might have been taken before the attenuation was changed on the board. If this doesn't fix it, I can modify the code to do a binary search or something instead of trying to calculate the attenuation.

baileyji, 2018-09-13:
@neelay893 Put in counter to fix.

commit 9bf82b75 (2018-08-26): changeAtten now returns after receiving acknowledgement from V7. Might fix #31
commit 5867245d (2018-12-12): Changed auto ADC atten to quit after 10 tries. Fixes #31.