MazinLab/MKIDReadout #55: PS GUI Improvements

neelay893, 2019-08-02:
1. The top plot isn't very useful, maybe switch back to what we had before (max IQV and 2nd max/max). At minimum, we need all values visible to make it easy to select attens.
2. When you click on a new power, frequency should not change
3. Would be useful to be able to zoom in to a finer frequency range; particularly for windows that have a lot of resonators.
4. Might be useful to have a "marginal" flag for resonators that are there but noisy/questionable

neelay893, 2019-11-18:
5. Lines marking out resonators on IQV plot should change be a distinct color if the user has already reviewed/set power. This would help keep track of doubles.
6. IQV plot scaling needs to be set wrt to data being plotted, not all IQV data for that resonator.
7. Widesweep data plot scaling is off, and should not overlap with IQV plot.
8. Add button for marking resonator bad.

zobristnicholas, 2019-11-20:
9. Allow adding/removing resonators in the current window (sometimes the code creates too many or misses some).
10. Color resonator markers differently for those that have already been optimized