MazinLab/MKIDReadout #56: High Templar Longsnap

neelay893, 2019-08-02:
Longsnap plotting in high templar is broken b/c of key errors in the config object. I'm using the latest develop branches of both mkidcore and mkidreadout. Seen on both darkness and mec.
1. When you try to switch plots:
Traceback (most recent call last):
  File "/mnt/data0/neelay/MkidDigitalReadout/mkidreadout/channelizer/RoachPlotWindow.py", line 152, in plotPhaseNoise
    self.makePhaseNoisePlot(**kwargs)
  File "/mnt/data0/neelay/MkidDigitalReadout/mkidreadout/channelizer/RoachPlotWindow.py", line 180, in makePhaseNoisePlot
    fftlen = self.config.get('r{}.nLongsnapFftSamples'.format(self.roachNum))
  File "/mnt/data0/neelay/MKIDCore/mkidcore/config.py", line 236, in get
    return self[key]
KeyError: 'nLongsnapFftSamples'
2. When you try to collect a long snap:
Traceback (most recent call last):
  File "/mnt/data0/neelay/MkidDigitalReadout/mkidreadout/channelizer/RoachStateMachine.py", line 679, in getPhaseStream
    hostip = self.config.templar_host_ip
2019-08-02 13:42:53,029 mkidreadout.channelizer.RoachStateMachine getPhaseStream: INFO r119:ch701 Collecting phase timestream 
  File "/mnt/data0/neelay/MKIDCore/mkidcore/config.py", line 165, in __getattr__
    return self[k1][krest] if krest else self[k1]
KeyError: 'templar_host_ip'

neelay893, 2019-08-02:
The second error was my fault; didn't have templar_host_ip in hightemplar.yml. I have 'nLongsnapFftSamples' as a key in roach.yml.

baileyji, 2019-08-03:
Hmm. Looking at the code the QT window class is being passed config.roaches. The function is trying to get a roach specific key r###.nLongsnapFftSamples from that config which would inherit from the roaches parent (where nLongsnapFftSamples is defined in the default conf). Might try setting a breakpoint and inspecting self.config to see what it contains. It behaves much like a set of nested dicts with inheritance and the ability to access keys as attributes.

neelay893, 2019-08-05:
Turns out it was b/c all of the config keys get converted to lowercase; both the config file and code had uppercase characters.

commit 14f2d11a (2019-08-05): Fixes #56
closed 2019-08-05 by neelay893