MazinLab/MKIDReadout #59: Dashboard Log doesn’t like numbers or new lines

steigersg, 2019-09-08:
It seems to only take in strings as input

baileyji, 2019-09-08:
What do you mean? The comment field?

steigersg, 2019-09-08:
yes

baileyji, 2019-09-08:
The field is a PyQt4.QtGui.QTextEdit() And is read with with the toPlainText() method. It is converted to json for with json.dumps so maybe its not that the field won't accept it but that it causes a problem down the line. 

If that looks like it is more the case, you can try a live fix I'd suggest looking at swapping
`str(self.textbox_log.toPlainText())` in state() on ~1065 on line for `str(self.textbox_log.toPlainText()).encode('utf8')` ore the like. What I'm thinking is that the text needs to be escaped properly before getting dumped into json by dumps later. I'd thought dumps auto-escaped and handled all of unicode. Maybe I didn't RTD closely enough.