MazinLab/MKIDReadout #64: Threading error in dashboard

nswimmer, 2019-10-20:
After dashboard has been running for a long time with the black fridge, an error comes up where a new thread cannot be started. See traceback below

[image]

nswimmer, 2019-10-20:
When this error appears, data is still being taken and written to the data directory, I think the problem it has is with the number of threads it has made from creating FITS files. Not sure if this is because this is the first time we've taken data for this long and so it hasn't been a problem before, but it doesn't seem to be messing up the data itself.

baileyji, 2019-10-21:
I've done some work here. Time to test it again.

closed 2019-10-24 by baileyji